jasonsbeer/Amiga-2000-EATX #3: RTC-62423 accuracy is likely much worse than the original design

darrenfreeman, 2021-07-29:
The RTC-62423 datasheet specifies the frequency accuracy to be 20 ppm for the 62423A version, and 50 ppm for the 62423. This is really lousy compared to what you can achieve using a trimmer capacitor and a frequency counter. It might be even worse if the chip is decades old.

To put this into perspective, a 1 ppm error is one second over 11.6 days. So the non-A version could be out by one minute per fortnight. If you're using your A2000 only occasionally, it will be really noticeable every time you go to use it. And there's no way to adjust that.

I'm sure Commodore would have trimmed them all in the factory. This substitute is really a step down from the original hardware. If you need access to a frequency counter, go to an amateur radio club and ask nicely, someone will be happy to trim it to <1 ppm for you. (And now you have a new friend.) Or you could monitor whether it is running fast or slow compared to other clocks, and tweak it as needed.

Could you please include the component footprints needed for the capacitors and watch crystal, for those who want to use a part with a trimmable external oscillator? I am looking at a "parts only" A501 on eBay, and thinking that a through-hole footprint for the chip would be a helpful addition, alongside the SMD version.

jasonsbeer, 2021-07-30:
Adding worst case scenario notes for future reference...

62423A may lose just over 10 minutes per year
62423 may lose just over 26 minutes per year

darrenfreeman, 2021-07-31:
A DIP footprint would also be helpful if someone comes up with a replacement, e.g. a modern RTC with some translation to the 4-bit interface.

darrenfreeman, 2021-07-31:
BTW I really admire what you've done, I can't wait to build one.

jasonsbeer, 2021-08-02:
> BTW I really admire what you've done, I can't wait to build one.

Thanks!

jasonsbeer, 2021-09-26:
Revision 2.0 now includes the option of installing the original Amiga RTC circuit with the MSM6242 or using the RTC62423 design, which is simpler, but less accurate.

closed 2021-09-26 by jasonsbeer