jasonsbeer/Amiga-2000-EATX #7: Bom and CPL files for components

kipper2k, 2021-08-13:
Hi, Great work, i would like to give this board a build but TBH i don't want to fiddle with the small components. Is there a BOM and CPL file with the gerbers available so i could get the board partially assembled. Without the .Brd file i cant generate the X and Y coordinates, the only way i could do it is to put in gerber viewer and manually do it (Ouch). Even basic CPL with the coords listed would be a headstart. If i generated the assembly files i would upload to this project page for others to use. 

thanks  Brian (Kipper2k)

Acill, 2021-08-13:
I generated a CPL file, but I cant get the BOM to generate correctly. Here
is the CPL. I only edited the headers for jlcpcb on the top file. You will
need to open the top and look at headers and rename them for bottom if you
dont want to hand assemble the few parts on the bottom yourself. IF anyone
can get a BOM that works at jlcpcb please let me know as well!

-Paul (Acill) Rezendes

kipper2k, 2021-08-13:
Thanks for the reply. The BOM is not that bad to do for JLC, it is just very picky with upper case, spaces and commas etc. if there is a mistake then the BOM generation just stops without any clue as to why.  Did you post the CPL, i cant see it ?

Acill, 2021-08-13:
I was able to do the BOM.

Acill, 2021-08-13:
Seems I can post the files. If Jason has it set for no uploads from outside
users. If you email me at ***@***.*** I can send you them.

-Paul (Acill)

kipper2k, 2021-08-13:
hi, can you send to kipper2k@shaw.ca

thanks

jasonsbeer, 2021-08-13:
Hi everyone....I'm not sure why you can not drop files in your replies. I have not intentionally blocked it. In fact, others have posted files to other issues. I can look into it later when I get home.

jasonsbeer, 2021-08-14:
I don't see how to affect changes to the upload settings for the issues tab. If you are unable to attach files, maybe you have an idea of what to look at and I can follow up. I would very much like to include the files you are working on for PCB assembly. I'm sure others will find them useful. 

Thanks

mikebetz42, 2021-09-01:
> Seems I can post the files. If Jason has it set for no uploads from outside users. If you email me at ***@***.*** I can send you them.
> […](#)

Could you send me a copy of the files please?

mikebetz42 at hotmail dot com

Thank you,

Mike

jasonsbeer, 2021-11-15:
Closing this due to lack of discussion for some time.

closed 2021-11-15 by jasonsbeer