jasonsbeer/Amiga-2000-EATX #10: Questions about pcb

nobodyisinocent, 2021-08-15:
Hello,

I have downloaded the Kicad project. What you have done is really great !
There are some holes which are not connected like the Int6 signal onto the Zorro slots.
The 3v3 pins of the power connector are not connected also.

Is it normal that there are some missing tracks  as shown in the pictures below ?

Int6 signal:
[image]

3v3 signal:
[image]

Kind regards,
Franck

nobodyisinocent, 2021-08-15:
To be conform to the schematics, the Int6 holes should be connected. Something like that: :-)

[image]

nobodyisinocent, 2021-08-15:
I assume you used Freerouting for generating the tracks.
This software is excellent, but the optimization batch can have some weaknesses.
Like the original tracks dispositions on CN600 for example. On the left the automatic rooting, and on the right a manual modification of the tracks.

[image]

I'm actually trying to optimize the pcb of your fantastic project ! :-)

nobodyisinocent, 2021-08-15:
A little work of rooting optimization around U102.
Here, the original rooting:
[image]

Here, the modified rooting:
[image]

nobodyisinocent, 2021-08-15:
Around beautiful ANGUS ! :-)
Before:
[image]

After:
[image]

jasonsbeer, 2021-08-15:
The _INT6 signal is connected correctly in the supplied files. Perhaps something was disconnected during your optimization work.

[image]

The 3.3V taps don't need to be tied together. I probably should have removed them from the schematic, but chose to ignore them instead. Only X2 uses 3.3V and that's connected properly. So, it is as expected.

The board is completely routed by hand. I did not take the time to review everything for optimization. My thought, if it works, let it be. I'll look at your suggestions for consideration in a possible future revision. Thanks

closed 2021-08-17 by jasonsbeer