jasonsbeer/Amiga-2000-EATX #13: 8375 Agnus Support

jasonsbeer, 2021-08-21:
Paging the Amiga RAM team! @texneus @LIV2

I desire to make an alternate revision of the Amiga 2000 EATX that directly supports the 8375 2MB Agnus. I have attached a proposed schematic of the RAM circuit. Not everything is connected, but all the parts relevant to this discussion should be. Can I impose upon you guys to review the schematic and let me know what you think? 

Thanks!

[CY62167E_16_Mbit_1_ M_ 16_Static_RAM.pdf](https://github.com/jasonsbeer/Amiga-2000-ATX/files/7024724/CY62167E_16_Mbit_1_.M_.16_Static_RAM.pdf)

[file6121153C-Sheet6121153C.pdf](https://github.com/jasonsbeer/Amiga-2000-ATX/files/7024725/file6121153C-Sheet6121153C.pdf)

LIV2, 2021-08-21:
No worries @jasonsbeer I've had a look and it looks good to me!

jasonsbeer, 2021-08-21:
> No worries @jasonsbeer I've had a look and it looks good to me!

Thanks, Matt!

darrenfreeman, 2021-08-21:
Will it be backwards compatible with a 1 MB Agnus, even with the 2 MB SRAM? I.e. will the uppermost address line to the SRAM be a constant during access?

If so, you could build a 2 MB board, and then go searching for the 2 MB Agnus sometime later. This is especially relevant if someone makes up a batch of boards without the custom chips.

If not, would it be hard to add a jumper for forcing it to only see 1 MB of the SRAM?

jasonsbeer, 2021-08-21:
> Will it be backwards compatible with a 1 MB Agnus, even with the 2 MB SRAM? I.e. will the uppermost address line to the SRAM be a constant during access?
>
> If so, you could build a 2 MB board, and [...]

I have been thinking a lot about this concept. Unfortunately, the 8375 is not pin compatible with the 8372A. By quick count, there are 21 signals on a different pin. Using a single board to support both versions would require a considerably large field of jumpers and a mess of traces to make it work. This would take a lot of rework to implement, but I certainly agree a single board approach is preferable. I will see what I can do.

jasonsbeer, 2021-08-21:
In order to explore a single board to support both 8372A and 8375...I worked up the schematics. I ended up with 34 jumpers that would need to be configured to support either configuration. 21 to configure Agnus, and 12 to configure the SRAM. I call it the Insane Jumper Edition. I'll upload it to the dev branch if anyone gets a desire to check it out. I do not intend to follow that path any further at this time.

darrenfreeman, 2021-08-22:
What about two sockets? Could you make enough room to place them side by side, and you only populate one?

LIV2, 2021-08-22:
It might be possible to fit two sockets, perhaps if the footprints were overlapping but offset diagonally by 1.27MM (kinda like I did on my A500 8375 adapter).
I wouldn't recommend this though, adding dual sockets will result in stubs for every signal going to Agnus which can result in instability and radiated noise.

darrenfreeman, 2021-08-23:
Stubs won't be an issue at this clock frequency, they will just add a tiny lumped capacitance. (The capacitance of the traces on the four-layer ATX version will be completely different to the original.) The Agnus is 30 years old, and the original B2000 board was two layers and used 1 kOhm terminations.

The two-socket PCB would be an alternative to the Mini Megi style of adaptor, and those ran an address line over a twisted pair.

Here's one for comparison, on a real rev 6.2 board: https://www.ebay.com.au/itm/203532038100

jasonsbeer, 2021-08-24:
I am working with @texneus on this. We have a couple options we are evaluating. Stay tuned.

texneus, 2021-08-24:
I'm going to reopen this as I think there are ways to eliminate jumpers that are worthy of further discussion, and personally I would like to see this as one board layout for a variety of reasons including those already discussed.  Focusing on 2Mb SRAM only and spring boarding of some schematics created earlier by @jasonsbeer, the following gets us down to "only" 14 jumpers and (assuming my eyes didn't get crossed somewhere) this should work with either Agnus.

This particular design takes advantage of the fact that most of the Agnus RAM address lines (DRA0-9) are still address signals even though they are not in the same place between Agnus chips.  In the end, the SRAM won't care as long as it's addressed consistently.  In 1Mb mode A0 is logical 0 at all times, effectively allowing only half of the 2Mb SRAM to be addressed.  The other addresses lines would skewed as shown on the second sheet.  For 2Mb all address lines line up.  Feel free to review and point out the error of my ways.

Also, at least one jumper (at the bottom of the first sheet) is redundant and can be removed.  The silk screen could be changed to "Short for NTSC or 8375 Agnus".

Please excuse the crudity of this drawings, I did not have time to paint it or build it to scale...I will learn KiCAD at some point, but until then I will do layouts like its 1989 :-)

I have not looked at other jumpers yet but there are probably opportunities to reduce these further as well.

[image]
[image]

darrenfreeman, 2021-08-24:
On my A500, I remember doing a mod for 1 MB chip instead of 512k chip and 512k pseudo-fast. IIRC that was a jumper implemented as three pads and a track between two of them, I had to cut the track and bridge the other two. (Forgive me if that's not quite right, it was in the early '90s!)

Such an approach, defaulting to the 1 MB Agnus, would make almost everyone happy. And it would be period-correct! :)

darrenfreeman, 2021-08-24:
I also wonder, in relation to that 512k chip and 512k pseudo-fast. Was this done to allow boards with the older Agnus to address 512k of chip, while allowing the CPU to address the rest? Can this approach work with 2 MB of SRAM, and a 1 MB Agnus?

Or was it only ever possible to do that with a 1 MB Agnus, and they downgraded it for other reasons?

LIV2, 2021-08-24:
@texneus that's clever! Very nice

> I also wonder, in relation to that 512k chip and 512k pseudo-fast. Was this done to allow boards with the older Agnus to address 512k of chip, while allowing the CPU to address the rest? 

Yes the older Agnus' internal registers could only make use of 512K Chip but the memory controller portion could address up to 1MB which allows the CPU access the slow ram 

Presumably even when they came out with the 1MB Agnus they left it defaulting to 512K chip / 512K slow to ensure compatibility with existing software that might have assumed that configuration

> Can this approach work with 2 MB of SRAM, and a 1 MB Agnus?

Unfortunately not. 1MB Agnus only has enough address inputs for 1MB so the memory controller can't address more than that.
With a bunch of external logic

darrenfreeman, 2021-08-24:
Oh well. So, how about a jumpered configuration as per @texneus, but done using 0603 0R resistors? And including a default configuration of 1 MB, implemented using thin tracks that will be easy to cut.

If, at some point in the future, you're fortunate enough to get a 2 MB Agnus, you just install the 0R resistors (or solder blobs) across the pads that are not shorted, and then cut those tracks. It should be easy enough to get the resistors on the right pads, by making sure you can still see the tracks that you need to cut.

jasonsbeer, 2021-08-24:
Just to add some info to the schematics texneus uploaded. I drew up a symbol that would cover both Agnus versions in question (8375 and 8372A). On the symbol, you will see some pins identified with slashes in the name. These are (probably obviously) the pins that differ between the two versions of Agnus. The format is [8375 Signal]/[8372A Signal].

You guys can double check I drew it up correctly, as well.

darrenfreeman, 2021-09-04:
This was news to me - there are (at least) two different pin-outs for the 8375, depending on the CBM part number. It will not be sufficient to call the chip "8375". That designation should be dropped, going forwards, as baffling and inconvenient as this is.

https://www.amigawiki.org/lib/exe/fetch.php?media=de:parts:agnus-history.pdf

From my reading, the A3000 used a 2 MB Agnus, with a similar pinout to the 1 MB Agnus of the A500/A2000. The A500 Rev 8a, A500+, and A600, use the modified pinout that requires all these jumpers.

I think we can support all the 1 and 2 MB parts, it will just require a table of jumper settings for each CBM part number. Perhaps it can go on the bottom silk screen, with all the jumpers.

LIV2, 2021-09-04:
The 2MB A3000 Agnus is practically unobtainable though

darrenfreeman, 2021-09-04:
According to this thread, the A500 Rev 8a had the A3000 version, while the A500+ (same motherboard, different components) had the A600 version.

http://eab.abime.net/showthread.php?t=52072

So it will be interesting to look at the schematic for the Rev 8a, maybe we will find a lot of jumpers...

darrenfreeman, 2021-09-04:
I had a look, couldn't see anything related to that. Maybe the above thread is mistaken. Maybe the board didn't work, because someone put the wrong part on it.

jasonsbeer, 2021-09-10:
Here is a list of 8375 part numbers that I think we can support without going totally over the top. I would have to tweek existing jumpers, but not add any. Part 390551-01 is what I have been using on the schematics up to now.

|Part Number|PAL/NTSC|Comment|
|-|-|-|
|390551-01|PAL|8375R0 Original Release|
|390551-02|NTSC|8375R0 Original Release|
|318069-12|PAL|Differs from 390551-01, pin 41 is Vbb instead of GND|
|318069-13|NTSC|Differs from 390551-02, pin 41 is Vbb instead of GND|
|318069-10|PAL|Differs from 390551-01, pin 40 is Vbb instead of 14MHz|
|318069-11|NTSC|Differs from 390551-02, pin 40 is Vbb instead of 14MHz|
|390544-01|PAL|Pin compatible with 318069-10|
|390544-02|NTSC|Pin compatible with 318069-11|

This information is taken from https://www.amigawiki.org/lib/exe/fetch.php?media=de:parts:agnusliste_v09_120923.pdf, which has provenance. Anyone reading this is welcome to review this list. Let me know if you see any discrepancies.

jasonsbeer, 2021-09-20:
Good discussion here, everyone. I think this thread has gone as far as it could. The new board revision is routed and uses solder pad jumpers to configure the board for the Agni shown in my previous post.

closed 2021-09-20 by jasonsbeer