jasonsbeer/Amiga-2000-EATX #14: Enhancements to the internal floppy connector

darrenfreeman, 2021-09-03:
I have been comparing the A500 and B2000 internal floppy connector circuits. The A500 only supports one internal drive, because it turns the motor on only when DF0 is meant to spin. (Incidentally, can we agree that _MTROD is a typo? It's called _MTRON on the A500.)

The B2000 drives the _INUSEx lines as you would expect the motors, accounting for the twist in the cable before DF0. Then it logically ORs those into _MTRON, so both motors spin if either drive is in use, it's just the LEDs that differ. (Ouch, my antique floppies, they will wear out faster!) Despite the cable twist, the drives still need jumpers set for which _SELx line they respond to.

Since we are aiming at a 100% new parts build, I figure I'll use PC drives. Either modded to work like Amiga drives, or using an adapter right at the drive. I'd do a bit more twisting on the cable, so both drives only have to listen to one of the _SELx lines, and they become interchangeable.

Then you can modify the drive to spin the motor when it is _INUSE, which restores the behaviour of external drives, i.e. all drive motors are independent. The common _MTRON signal can be dropped.

And since ATX cases usually have two bays, I'll install two drives. But then I'd also like a Gotek, hmm... and there's no external drive port any more...

One approach is to just bring out _SEL2 and _SEL3, as well as _MTRx and _DKRST, so the internal floppy port has the needed signals to supply drives using external floppy controllers. But, it's the same number of pins if we supply _SEL2, _SEL3, _INUSE2, _INUSE3, and that would make the case layout so much neater. All drives are treated the same.

So, I suggest that instead of one jumper, for DF1, there would be three. And a couple more logic chips to derive the extra four signals.

You can free up pins 14 and 16 straight away. Make them _SEL2 and _SEL3, respectively. It's a neat solution, the cable shifts the wires by two places after each drive. (Pull out the sharp metal part of the IDC connector, for any new signals that you don't want to send to a given drive.)

And pin 3 is usually cut off, but it could be used. Perhaps as an _INUSEx, because those signals don't really need twisting with ground anyway. E.g. wire the motherboard as:

* 4 - _INUSE0
* 5 - _INUSE1
* 6 - _INUSE2
* 7 - _INUSE3

Similar to above, the wires move one across after each drive.

As I've specified the above pins, the drive closest to the motherboard would be DF0. This means you can make a cable for only one drive, without any twists in it. If you want DF0 at the far end of the cable, you twist the _SELx and the _INUSEx groups, between the motherboard and the first drive. The drive ordering is now reversed, with DF3 being closest to the motherboard.

I think that's pretty close to optimal, really. Simple cable wiring, take some photos and anyone can copy it. It does mean changing the drive behaviour, so the motor spins on _INUSE. (You can do this in the cable, if you really want. But if you're modding PC drives anyway, what's another wire.)

darrenfreeman, 2021-09-08:
I see your point about EATX cases not supporting floppy bays, but there are plenty of brand new ATX cases that do. I don't have an opportunity to look at cases in person, due to the pandemic, so I can only be optimistic about finding one that can be made to fit the EATX board. I still have a roomy ATX case with 4x 5.25" and 2x floppy bays, and I can get more.

I'm not personally interested in modernising everything about the system, if that means losing functionality. I think there will be builders who feel the same way. I'm interested in trying out things I never got to try 30 years ago, things you can't do in UAE.

I think I could fill up all four floppy drives. For example, 2x 3.5" floppies (so you can X-copy), one GoTek drive, and one 5.25" drive for reading even older discs (e.g. C-64).

Here's another suggestion: provide the external floppy connector as a 26-way header on the motherboard, using the pinout that would allow a ribbon cable to be soldered to a DB23 female connector. These are now available for a reasonable price (under $10), so some builders will want to install one on the back panel.

Then it would be pretty easy to plug a small adapter board into that 26-way header, which provides the logic to drive a secondary 34-way internal floppy connector. Those who want more internal drives, could get them with a second standard A2000 floppy cable.

neilo40, 2021-09-08:
> Intriguing idea. Do you feel there is a good use case for more than 2 floppy drives? I'm not sure it fits into the forward looking/modernization intent of this project. Projects like WHDLoad have greatly decreased their usefulness. External drive bays are disappearing from all computer cases.
> 
> Alternatively, I would like to consider a future project or enhancement that brings removable USB drive compatibility to the platform.

I really like this idea.  I'm imagining a gotek-like capability on the mainboard with the USB port interface via standard header so that a case USB port could be used.  control and feedback would need to be thought through, but +1 for this direction

darrenfreeman, 2021-09-08:
If you want a USB floppy emulator, you can just use the floppy interface as it stands. Emulators are an off-the-shelf item. Note, I was originally asking if we can have four drives internal, because I want two floppies _and_ an emulator. But I also like the idea of mounting a DB23 on the back of the case, since those are available. Then I can plug in whatever devices I feel like, at the time. And as I suggested, that header can be used as a second copy of the internal header, if that's what you want. (Just need a daughter board with a couple of chips on it.)

I would certainly like to have native USB support though. But there are Zorro cards that offer this, and what's the point of a Zorro backplane if you stuff everything into the motherboard anyway ;)

For games, you need to run the mouse through the custom chips. There are USB to mouse/joystick adapters already, as off the shelf items. But what would be pretty nice, is if we could fit them internally and use the USB sockets on the front of the case. Perhaps add two internal mouse/joystick headers, which combine the signals with the external ports?

I'm not sure what happens when you unplug the USB mouse from the various adapters, the ideal behaviour is they let all signals float high. So you just wire it in parallel with the external mouse port, and use whichever one you want at the time.

It would suck if they hold two of the quadrature signals low, as then you need a way to isolate them in order to use an Amiga mouse/joystick. It would require a bus mux and a toggle switch, to decide whether to use the external port, or the internal pin header. Quite doable still, but if anyone can confirm there is a USB adapter that goes completely passive when you unplug the USB device, then it's a really easy thing to add them in parallel with the external Amiga ports. Adds a few cents to the board, for a whole lot of options.

jasonsbeer, 2021-09-26:
Revision 2.0 includes an internal header that supplies the external floppy signals. The user can build a DB23 adapter cable to support external floppies or otherwise tap into those signals as they choose.

closed 2021-09-26 by jasonsbeer