jasonsbeer/Amiga-2000-EATX #15: EMI filters

Pragakhan09, 2021-09-07:
Hi, 
first of all, @jasonsbeer thank you for this wonderful project ! I'm going to build one but I see that those EMI filters (W3F11A1018AT1F) are not available. Can someone suggest a 1206 form factor alternative ? I'm wondering if the 0805 form factor ‎101X15N101MV4E‎ would fit ?
Kind regards

jasonsbeer, 2021-09-08:
If you can't find the exact part, you can try part NFE31PT101C1E9L. Its a bit odd looking, but is the correct footprint.

I would avoid using the 0805 part. You can probably get it to work if you place just perfect, but are likely to have trouble with consistent solder joints.

Pragakhan09, 2021-09-08:
Great, thanks for your answer. I looked at that part but like you say it's a bit odd looking and has a few different specs (25v...). I'll implement those :-)

closed 2021-09-10 by jasonsbeer