jasonsbeer/Amiga-2000-EATX #16: Looking to borrow or buy certain 8375 Agnus parts

jasonsbeer, 2021-09-20:
I have finished routing Rev 2.0, which is intended to support both 8372A and certain 8375 Agnus chips on the same board. I'll probably send this off for fabrication soon.

My next quest is to either buy or borrow some different 8375 Agnus part numbers. Please let me know if you can help out with any of the Agnus part numbers below.

390554-01
390554-02
318069-12
318069-13
318069-10
318069-11
390544-01
390544-02

I also added a discord channel for this project...https://discord.gg/NU7SPYfNFj

jasonsbeer, 2021-09-26:
> I was thinking of adding a wish-list issue for the next release. I intend to make five boards and distribute them locally, and I'm holding off to see how much of my wish-list I can get into the board.
> 
> For me, it would be:
> 
> * support for SOIC and PDIP RTC chips, and for chips with and without external oscillators
> * support for all four floppy drives. (Just add a header for the external floppy port, problem solved.)
> * internal headers for the two mouse/joystick ports, wired in parallel with the external ports.
> 
> Others may have items to add, so a dedicated issue might be worthwhile.

I added the option to use either the original Amiga RTC design with the MSM6242 or the RTC62423, as was originally included in this project. The user can decide which they prefer to install.

I added an internal header to support external floppies. The user can build a cable to DB23 and mount it to the case or tap into the signals however they choose.

For the mouse/joystick headers...the term "diminishing returns" comes to mind...in terms of time, PCB real estate, and general need. I don't believe this is needed for the large majority of users. It takes a lot of time to modify the schematics and re-work the board. If this request is to support an open source Amiga project, I will consider adding it. Otherwise, I'm at feature lock for this revision.

kinmami, 2022-04-03:
Hello @jasonsbeer
I have the following chips in stock pulled off from operating machines:

Agnus (1MB)	8375 R1			318069-16	PAL
Agnus (2MB)	8375 R2			318069-10	PAL
Agnus (2MB)	8375			318069-11	NTSC
Agnus (2MB)	8375			390544-01       PAL
Agnus (2MB)	8375 R0			390544-01	(should be PAL)
Agnus (2MB)	8375			390544-02	NTSC

I do prefer keeping the PAL versions, but we can find an agreement for the NTSC chips 318069-11 & 390544-02.
I am going to pvt you on the Discord channel.

closed 2022-04-12 by jasonsbeer