jasonsbeer/Amiga-2000-EATX #21: Flickerfixers and scandoublers do not work correctly in videoport

xtpower, 2021-11-18:
I have ben having some trouble using my 2 flickerfixers / scandoublers on the board.
A little backstory.
I have 2 scandoublers / flickerfixers which both works out well on my stock A2000 rev6

I have ben comparing the picture and signals from my org. A2000 and the A2000ATX board on my bench - I will post the pictures.
I'm writing because I'm at a dead end in this.

My setup is the same on both boards: Kick 1.3 - up and running, but not booted into anything.
I'm using a Lighttech Flickerfixer.
My second Flickerfixer is a microway: https://bigbookofamigahardware.com/bboah/product.aspx?id=374
This part do not work in the port for some reason.
Both machines are PAL machines, and the Xtal is 28,37516 MHZ

And here's some pictures of a simultaneous comparison of the videoslot.
The Burst signal on both looks like this
[image]
:

The hsync signal looks like this:
[image]

When on kickscreen, I got this:
Original A2000
[image]

ATX amiga
[image]

It is noticeable that there is a white barre just below the kick picture. 

Second: When trying to change screenmode to interlaced, a thing the flicker card can handle, the ATX board doe not handle this at all.
Two pictures to show this.:
Org amiga 2000:
[image]

ATX amiga:
[image]

Also a part of the picture (topmost) is out of the picture.

I do not know what causes this issue. All the signals is there, but they seems wrong in some way.
Maybe a misconfiguration of the videoengine??
Idunno.

jasonsbeer, 2021-11-19:
Tagging this thread as help wanted, for now.

I looked up the two cards you mention, and I found schematics for the Light Tech card. I suggest we concentrate on that one since the schematics and VDHL code is available. Please confirm this is the one you are working with: [https://www.lighttech-hg.de/index.php/open-source-projects/a2000-scandoubler](url)

The first thing that comes to mind is a bad clock somewhere. As a reference, I have been using the Pi -> HDMI video card in both PAL and NTSC machines without trouble, but I have no "flicker fixer" cards to test.

Let's start easy and work our way to hard...

It appears the Light Tech card uses the XCLK/_XCLKEN signals. Cards in the video slot use these signals to replace the system clock supplied by X1. If one or both of those signals are not functioning as intended, all the clocks will be running from X1. In that event, this would not correct for the hardware in the video slot.

I recommend you...

1) Check that _XCLKEN is LOW when your card is in. It should be HIGH when the slot is empty.
2) Probe the XCLK signal. What is the frequency?
3) Probe the 7M signal at pin 15 of the 68000. What is the frequency?
4) Probe the other clocks used by the Light Tech card..._VC3, _VC1, _VC4, _CDAC. What are the frequencies?

Let's go from there.

darrenfreeman, 2021-11-19:
@xtpower unplug anything that might be plugged into the SVGA connector, see if that helps.

xtpower, 2021-11-19:
Hi again.
I can confirm that the card I mentioned is indeed the card you are linking to.
I will have a look tonight reading the frequencies.

And yes, There's nothing plugged into the SVGA connector, so it's not that.

xtpower, 2021-11-24:
Just to come around your original question,
The frequencies is as expected.
7,09 and 3.54 on the testingpoints.

The _XCLKEN  is the same on both machines and the same with XCLK 

I have ben able to replicate the FlickerFixer issue,  (Picture of the Logica Diagnostic not using the whole screen) on My Amiga 2000.
Moving J300 Where you are choosing between _VSYNC and TICK.
On Amiga 2000 Rev 6. with position 1-2, this thing / issue where the screen is not filling out the full potential, vertically, occurs.

I guess that Pin 14 on CN207 still needs the original 50HZ no matter what the rest of the machine is up to.

I do have that raspberry pie HDMI card for the Video port, and it works despite all I have said.
I know that there is an ATX to A2000 power plug, containing a PIC that is producing a 50HZ ([LINK](https://www.ebay.com/itm/403251225065)).
I was wondering if a similar thing could be integrated on the board.

jasonsbeer, 2021-11-25:
>Moving J300 Where you are choosing between _VSYNC and TICK. On Amiga 2000 Rev 6. with position 2-3, this thing / issue where the screen is not filling out the full potential, vertically, occurs.
 
 >I guess that Pin 14 on CN207 still needs the original 50HZ no matter what the rest of the machine is up to.

At last it seems we have a definitive answer to why TICK was driven by an alternate source in the big box Amiga. HSYNC needs to be solid 50 or 60 Hz, and any hardware that overrides the system clock with also cause HSYNC to deviate from expected. VSYNC would also be affected. None of the wedge Amigas had a power supply TICK and would have suffered from this same issue. Fascinating. 
 
> I know that there is an ATX to A2000 power plug, that is producing a 50HZ ([LINK](https://www.pcbway.com/project/shareproject/Tick_signal_generator_for_Commodore_Amiga_2000__3000__4000_Power_Supply_Unit.html)). I was wondering if a similar thing could be integrated on the board.

I'm not sure that specific example could be incorporated, as it needs connection to mains, and that is not supplied through the ATX power connector. There are other examples where programmable oscillators or PICs are used to supply the correct signal. I'll dig into this and see what can be done. In the meantime, I'll add this as a known issue so people are aware. 

Thanks for sharing your findings.

darrenfreeman, 2021-11-25:
That may not be the reason why they chose the mains frequency for the ticks. For a long time, the mains was used to set the timing in mains-powered clocks, and it was very reliable. Probably a more accurate wall-clock time than the crystal oscillator on the Amiga motherboard.

The wedge Amigas don't have a tick signal on the connector, so it's not an option for them.

I feel like this specific issue is more of an accident. Although the truth is that unless you're genlocking to a signal for broadcast, it's possibly even worse for measuring wall-clock time. A VCR in playback for example... all bets are off :D

darrenfreeman, 2021-11-25:
I was hoping to bodge a connector onto the board, so that I can inject my own tick signal. Perhaps that can go into the next revision? Perhaps make it a three pin jumper as before, only pin 1 is ground. So you jumper 2-3 for internal oscillator, or you install a two-wire header onto pins 1 and 2, to supply your own tick.

jasonsbeer, 2021-11-25:
I did some brief investigation last night and found a tick circuit on aminet. I like this concept because it uses off the shelf parts that do not need mains and does not need programming. I will investigate if it is a robust solution. Any comments are appreciated.

http://aminet.net/package/docs/hard/Mix_and_Tick

In a nutshell, it's a 74HCT4060 ripple counter driven by a crystal that feeds into back to back flip flops. It will generate 50 or 60 Hz depending on the crystal frequency. My concern is how accurate and consistent the output frequency is.

jasonsbeer, 2021-11-25:
> I was hoping to bodge a connector onto the board, so that I can inject my own tick signal. Perhaps that can go into the next revision? Perhaps make it a three pin jumper as before, only pin 1 is [...]

> 

Yup. However we end up with a new TICK, we will make it jumper selectable like the original.

darrenfreeman, 2021-11-25:
Please don't put something cheap and nasty on the board, just make it a header for something that's as good as the user wants. I plan to use the RTC with trimmer option, and I will trim it as close as I can. I could see myself eventually making an accurate 50 Hz clock, for the system to keep proper time when it's on.

xtpower, 2021-11-25:
It took me a little while to find but here it is.
[Atx to a2000 adapter with tick](https://www.amibay.com/forum/want-to-sell/hardware-for-sale/amiga/121610-amiga-2000-3000-atx-power-adapter#post2399647)
This may do the job, but be aware, it only do 50 Hz

jasonsbeer, 2021-12-02:
I pushed revision 2.2 schematics and Kicad drawings to the DEV branch. This revision adds a "tick header" that one can use to attach their own TICK generator circuit. I plan to evaluate a couple options for a simple TICK generator that will be posted as part of this project. I would feel bad if I put the header on the board and did not offer something to put on it. The header is a 1x6 2.54mm pitch male pin header.

If you do not need a TICK signal, short pins 1 and 2 to use /VSYNC, which is the current Rev 2.1 ("A500") configuration.

TICK Header
Pin|Description
-|-
1|/VSYNC
2|TBASE (TICK)
3|GND
4|/RESET
5|NC
6|+5V

jasonsbeer, 2021-12-14:
A quick update...
I have prototyped two stand alone TICK circuits. One that @texneus devised, which is driven by a 9VAC* wall wart and another that is oscillator driven by Michael Schulz (referenced above from Aminet). First impressions of both are very positive and either should make a good substitute when a discrete TICK is required. Having the two options will allow users to select which they prefer. The KiCAD files for both have been uploaded to the TICK folder in the DEV branch.

*Other voltages are supported via resistor value

Here is the 9VAC TICK output. 60Hz, since I am in North America. 50Hz will work as well.
[image]

Here is the output from Michael Schulz's project. Both 50 and 60 Hz.
[image]

[image]

xtpower, 2022-01-19:
I have had the time to test out the 50/60HZ external solution, and it is working.
Normal Lores and Hires is working well, so there is a reason for the TICK jumper configuration on the original A2000.

jasonsbeer, 2022-01-19:
> 

This is great news. Thank you for helping with this. I am in the middle of building a Rev 2.2 board to test it before release. I had to wait for some parts and now things are busy. I hope to have it completed and pushed to main very soon.

xtpower, 2022-01-19:
Thanx, and great work.
I appreciate you taking things serious, and taking steps towards solutions.
This is a mighty fine project, and I have ben flashing it around :)
I guess This issue can be closed - for my part.

jasonsbeer, 2022-01-25:
Great! I finally have my 2.2 board running today. I will spend the next few days checking stability and the new TICK header. I very much appreciate your help and patience with this issue.

jasonsbeer, 2022-02-04:
Revision 2.2 with the TICK header (J300) and the two TICK generator options has been pushed to main.

closed 2022-02-04 by jasonsbeer