jasonsbeer/Amiga-2000-EATX #22: Crowdfunding of assembled boards

Sgw32, 2022-07-13:
Hi everyone,

I've made a calculation for production on PCBWay and seems like it is possible to make a small batch for a price of ~300 USD(all populated will be ~400 USD) per board. If anyone is interested we can make sort of crowdfunding. 
I'm located in Russia though so there can be some shipping issues because of all of these things happenning...  (EMS works worldwide though, ~70 USD shipping) but we can make a direct shipping from China if needed. If anyone if interested I think we can make a collective order and split parcel to small parcels with direct delivery. Payment is also a question, however we can discuss it here. I now have 7 customers in Russia so I'm almost ready to order 10 boards. Maybe someone else wants to organize the order(for a larger batch), I can pay for boards. 

jasonsbeer, sorry for bothering you with this message. I think that anyway collective procurement and crowdfunding is more efficient than separate ordering and will help more people to try the project. 
I can share assembly order files(BOM, Pos) if needed - it will help many people and save time&money. 
And of course thank you for your amazing work, looks impressive!!!

BR,
Fedor

jasonsbeer, 2022-07-13:
Sounds like a great plan! You may to place this on the Discord channel.

jasonsbeer, 2022-09-07:
Closing due to inactivity for some time. Let me know if it should be reopened.

closed 2022-09-07 by jasonsbeer