jasonsbeer/Amiga-2000-EATX #33: Q: Will there be any chance of buying an ready board for us with disatreous soldering skills?

janderssonse, 2023-02-19:
This project looks amazing, I bet there are quite a few out there, including me, that don't hjave the skills,. nor the time free to learn the skills needed to succesfully order a board and make it a reality, i.e ready to put inot a case. I fully understand that you the creator rather perfect the design than to build pre-populated boards, but have any of the regular amiga dealers( the few that ar active) showed any interest in this.Or is there any other way to obtain a mostly complete board, or plans for a more user ready viersion?

jasonsbeer, 2023-02-20:
I recommend you visit the Discord channel and open a discussion there. There have been previous runs of "mostly assembled" boards that were started by members of the community. Might be some interest and doing another.

https://discord.gg/NU7SPYfNFj

I have not been approached by anyone about doing runs for commercial sales.

janderssonse, 2023-02-20:
Thanks for the quick response, will join! Cheers!

closed 2023-02-20 by jasonsbeer