jasonsbeer/Amiga-2000-EATX #38: wrong termination on Zorro II bus pin 96

LordZodiac76, 2023-11-12:
Hello Jason,

The termination from C675 on pin 96 must be on pin 95. The orginal schematic from A2000_R6 has no termination on pin 96 but C678 on pin 95.

cheers Marco

jasonsbeer, 2023-12-13:
Confirmed. Pin 96 (_EINT1) is unnecessarily terminated. Pin 95 (_GBG) is not terminated, but is in the Rev 6.x Amiga B2000 design.

**Issues/Observations**: 
1) I am not aware of reports of Zorro 2 bus issues, so it seems devices are working as expected.
2) Termination of _EINT1 is not of concern.
3) The _GBG signal is a way for Buster to signal all devices on the Zorro 2 bus that one of the Zorro 2 devices owns the bus. Lacking termination on _GBG could result in signal reflections during assertion. My personal speculation is this signal is not commonly used.

**Recommendation**: Include fix in next PCB revision. A bodge to fix this may be applied to Rev 3.0 and earlier boards, but is not recommended unless problems are encountered when DMA Zorro 2 devices are present.

jasonsbeer, 2024-06-02:
Fixed in Revision 3.1

closed 2024-06-02 by jasonsbeer