jasonsbeer/Amiga-2000-EATX #39: Polarity of C408 (-12V Rail)

Matz70, 2024-01-05:
Hi
I'm in the middle of the build and I've just gotten to the point where I've put voltage on the mainboard for the first time to take some measurements regarding the power rails. That's were i have noticed that the voltage on C408 (-12V) is reversed. C408 is not populated yet, so it isn't a problem. But i'am a bit hasitant to put it in like marked on the silkscreen. C412 (-5v) is marked the other (correct) way. Am i wrong ?

[image]

jasonsbeer, 2024-01-06:
Confirmed. The polarity of C408 is incorrect. You should install it reversed from the silkscreen.

**Boards affected**: Rev 3.0 and earlier.
**Recommendation**: Remove C408 and install with the polarity reversed from the silkscreen.

jasonsbeer, 2024-06-02:
Fixed in Revision 3.1

closed 2024-06-02 by jasonsbeer